ncd-io/ncd-red-ads1115 #1: Raspberry pi onboard i2c

macdroid53, 2018-07-25:
Should this work with the PI onboard i2c port?

When I select I2C and i2c-1 the  I2C Connection combobox is outlined in red and the error triangle never goes away.

zaphod-42, 2018-07-26:
That error typically implies that the "add" button wasn't clicked after i2c-1 was selected, I'll try to get a test set up today, but we have previously tested on the native I2C port on the rasbpi without issue

zaphod-42, 2018-07-26:
I just set it up and noticed an issue with the default values on the I2C Connection, go ahead and set the multiplexer drop downs to something, you can leave the "Use Multiplexer" option unchecked and it should be fine, I'll see if I can get this corrected today.

zaphod-42, 2018-07-30:
I've updated the dependency package to handle the default values appropriately. Closing

closed 2018-07-30 by zaphod-42