ncd-io/ncd-red-ads1115 #2: Always reading disconnected

rjackson159, 2018-07-26:
After configuring the ADS1115 node (using native I2C on rpi3) it constantly shows disconnected when I deploy my flow. I have an unofficial node that is not as configurable (node-red-contrib-ads1x15) and the ADS1115 board I'm using works fine with that node. I have iterated through all connection settings on hte ncd node. What could be causing it to not run with the ncd node but to work with an unofficial node? Also test.js won't run because of

 "this.wire = i2c.open(port, (err) => {

TypeError: i2c.open is not a function
    at new NcdI2C (/home/pi/node_modules/ncd-red-comm/lib/NcdI2C.js:12:19)
    at Object.<anonymous> (/home/pi/node_modules/ncd-red-ads1115/test.js:15:12)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3"

zaphod-42, 2018-07-26:
It looks like the i2c-bus library was not properly installed, you could try reinstalling with the unsafe-perms flag:

`npm install --unsafe-perm ncd-red-ads1115`

zaphod-42, 2018-07-26:
Did you set an interval on the ADS1115 settings? Without an interval it never calls out to the device, and so never knows if it is actually connected.

rjackson159, 2018-07-26:
Does the node require an input to function or can I use it strictly as an output? Currently I have been trying to configure it without an interval and use an input to initiate a reading.

zaphod-42, 2018-07-26:
You should be able to send a message with the topic of "get_status" to force a reading

rjackson159, 2018-07-26:
Okay, after I get all of my installations performed I'll update this thread. I suspect it was because I was not defining an interval or injecting the appropriate msg.payload

zaphod-42, 2018-07-26:
ok, though the msg.topic is currently what needs to be set, the payload is ignored (I'll probably improve this in the future, but will maintain the topic option for backwards compatibility)

rjackson159, 2018-07-26:
When I install ncd I receive these errors, could this be part of the problem? 

sudo npm install -g --unsafe-perm ncd-red-ads1115

> i2c-bus@3.2.0 install /usr/lib/node_modules/ncd-red-ads1115/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/usr/lib/node_modules/ncd-red-ads1115/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/usr/lib/node_modules/ncd-red-ads1115/node_modules/i2c-bus/build'

> serialport@6.2.2 install /usr/lib/node_modules/ncd-red-ads1115/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=8.11.3 runtime=node arch=arm platform=linux)
make: Entering directory '/usr/lib/node_modules/ncd-red-ads1115/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:95:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterUpdate(uv_work_t*)’:
../src/serialport.cpp:150:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterClose(uv_work_t*)’:
../src/serialport.cpp:188:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterFlush(uv_work_t*)’:
../src/serialport.cpp:231:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterSet(uv_work_t*)’:
../src/serialport.cpp:285:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterGet(uv_work_t*)’:
../src/serialport.cpp:336:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterGetBaudRate(uv_work_t*)’:
../src/serialport.cpp:383:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterDrain(uv_work_t*)’:
../src/serialport.cpp:424:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
../src/poller.cpp: In static member function ‘static void Poller::onData(uv_poll_t*, int, int)’:
../src/poller.cpp:69:29: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   obj->callback.Call(2, argv);
                             ^
In file included from ../src/poller.cpp:1:0:
../../nan/nan.h:1618:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
  CXX(target) Release/obj.target/serialport/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/usr/lib/node_modules/ncd-red-ads1115/node_modules/serialport/build'
+ ncd-red-ads1115@1.2.2
added 197 packages in 72.175s

rjackson159, 2018-07-26:
Despite the above errors, I added the interval setting and the node now says it is connected but doesn't appear to be giving any output; I have it hooked up to a simple gauge and chart from Dashboard.

zaphod-42, 2018-07-26:
it should be outputting 0 with nothing connected to it, if it says connected then it has been able to connect to the device and send out a payload.

rjackson159, 2018-07-26:
It is connected with a 0-24Vdc power supply supplying 2.0V currently. Does the node output in JSON? or is it a number?

zaphod-42, 2018-07-26:
the final channel outputs JSON, each channel before that is associated with a channel on the chip and outputs a numeric value.

rjackson159, 2018-07-26:
Oh! I am attempting to read the output channel because it is not currently setup to output all channels. This seems obvious with that information, I kept occasionally seeing json formatted text popup on my gauge

rjackson159, 2018-07-26:
Okay, still having trouble getting an output and it no longer shows as connected, NR has crashed and is throwing this error continually: 

(node:1006) UnhandledPromiseRejectionWarning: Error: EREMOTEIO: remote I/O error, write
(node:1006) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)

rjackson159, 2018-07-26:
I'm not sure what had happened but a system reboot fixed it all... Thanks for you're help!