ncd-io/ncd-red-ads1115 #4: Adress problem

CristeaPaul, 2018-10-31:
Hi there, i have an  ADS1115 but my adress on that chip is 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --

It is a ADS1115 ADC bord it is a fake one? or is not compatible whit this node contrib? 
on the ncd-ads1115 node are only 2 address 72 and 73 to select
--

zaphod-42, 2018-10-31:
The node address selection is showing decimal, 0x48 == 72

closed 2018-10-31 by zaphod-42

CristeaPaul, 2018-10-31:
:( the node is  disconnected, but the chip i2c address is show un command i2cdetect -y 1

zaphod-42, 2018-10-31:
You either need an interval set on the node, or you need to inject a payload with the topic "get_status", otherwise the node never calls out to the chip, and can't determine the connection status