ncd-io/ncd-red-ads1115 #5: cant output more then 2 channels

ingtvis22, 2018-11-30:
I try to read 4 different analog values with an ads1115 but i cant get it to work
when i select more than two single ended channels i do get extra outputs in the symbol but there is no value attached.
the second value is alway the last channel,  (1-2) (1-3)(2-4) or in my case (1-4) 
The value in the device-status, stated at the third connection in the schematic and gives

30-11-2018 14:19:23node: 7aea7656.b3d8f8
device_status : msg.payload : Object
{ channel_1: 32767, channel_4: 539 }

the second two connectons give no value
i hope some one can help me with this

Remon

[{"id":"a0fce694.465218","type":"debug","z":"cfbc225f.2188e8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":470,"y":20,"wires":[]},{"id":"7aea7656.b3d8f8","type":"debug","z":"cfbc225f.2188e8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":490,"y":140,"wires":[]},{"id":"8ed0cc37.9b1d3","type":"debug","z":"cfbc225f.2188e8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":500,"y":60,"wires":[]},{"id":"2413dadd.45c456","type":"debug","z":"cfbc225f.2188e8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":480,"y":200,"wires":[]},{"id":"9ef8d3fc.da4f8","type":"debug","z":"cfbc225f.2188e8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":510,"y":100,"wires":[]},{"id":"9f783f49.9d5ef","type":"ncd-ads1115","z":"cfbc225f.2188e8","name":"alanaloog","connection":"a5103eb4.76d77","addr":"72","interval":"1000","outputs":5,"output_all":true,"output_mult":1,"channel_1":"4","channel_2":"5","channel_3":"6","channel_4":"7","gain":2,"mode":1,"rate":4,"compMode":0,"compPol":0,"compLat":0,"compQueue":3,"delay":"100","x":240,"y":80,"wires":[["a0fce694.465218"],["8ed0cc37.9b1d3"],["9ef8d3fc.da4f8"],["7aea7656.b3d8f8"],["2413dadd.45c456"]]},{"id":"a5103eb4.76d77","type":"ncd-comm","z":"","name":"IkwadraatC","bus":"i2c-1","commType":"standard","addr":"0","useMux":false,"muxAddr":"112","muxPort":"0"}]

zaphod-42, 2018-11-30:
Thank you for the detailed report, I don't have the hardware in front of me to test this today, but I'll try to have it tested and fixed Monday

zaphod-42, 2018-11-30:
I take it back! I just took a look and found a scope error in the get_status function... I've just released v1.2.4, if you wouldn't mind updating to this newest version and trying again, I think it will fix your problem

zaphod-42, 2018-12-03:
Closing for now, feel free to reopen if you are still seeing this issue

closed 2018-12-03 by zaphod-42

ingtvis22, 2018-12-10:
finaly time to test it 
it works 
Thanks !!!