ncd-io/ncd-red-ads1115 #6: Says connected but I have no output

Barabba11, 2019-01-10:
Maybe it nees a "trigger" from the input? Which one should I send it? Anything from any output, I connected debug which work for other devices.

zaphod-42, 2019-01-10:
What platform are you on? Can you post an export fo the relevant nodes?

Barabba11, 2019-01-10:
THank you for kindly support me! I'm on raspberry 3, latest version of Raspbian, Red node, and your module
[{"id":"d12d0c64.a720e8","type":"inject","z":"fef64de3.66c588","name":"","topic":"","payload":"trigger","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":170,"y":400,"wires":[["fa8a38aa.82a28"]]}]

Barabba11, 2019-01-10:
may you kindly tell me if it needs any trigger at output to work? Any message inected?

zaphod-42, 2019-01-10:
You can either send a message with the topic "get_status" in, or set an interval.

Barabba11, 2019-01-10:
when I do that (topic "get_status") I see for 0,3 seconds a disconnected and then connected again. No output on any of the 5 outputs, never had. What can it be? How can I help you to debug?

zaphod-42, 2019-01-10:
I'm not really sure, are there any error messages in the shell? I've never had an ADS1115 with 74 as an address option since there is only 1 address pin

zaphod-42, 2019-01-10:
I've never had that issue with ncd-red-comm being "sticky", I just ran a test and when I uninstalled ncd-red-ads1115 it also removed ncd-red-comm. Try setting an interval on the ads1115 node instead of injecting and see if that works

Barabba11, 2019-01-10:
Dear friend, thank you for assisting me, it works, but is it not really clear why it didn't work before:
THe problem was interval, can't be 1.. please write next to it "ms" and advice people they may have no output if the conversion is such fast. 
Anyway I still have some dubts from the outputs values I'm having, I let you know

zaphod-42, 2019-01-10:
Glad it's working, I'll add the ms in the next revision (it is in the help text for the node, but I understand it could be easier to find). I'm going to close this ticket since you are getting output now, feel free to open another if you have any more issues.

closed 2019-01-10 by zaphod-42

Barabba11, 2019-01-16:
Hi there! I need again your help!
So as soon I connected a trimmer to every channel to test it I noticed that the module output channel 1 is connected to physical ADC3 input, and it displays about the same result (some time equal, some dime it differs from some units) to module output 4. Channels 2 and 3 output the same values, of channel ADC1.
This may be generated from wrong settings of MUX inside ADS1115, in your module I can configure address of mux and its port, I don't know what these datas are useful to, I've read in ADS1115 datasheet, the mux can be configured from config register as well as other functions existing in your module such comparator and so on, ADS1115 has not other I2C address. IN MUX is possible to choose I2C address  as 112 to 119, and port, is is an eventually external MUX too? In this case I've to ignore it, but the settings "single ended" where I suppose to configure the internal mux I have random effects, for example:
I enable all inputs "single end", I have low voltage on ADC3, and I see the output ar ch4. If I disable the channel 2 I don't see more low voltage pin, ch1,3,4 have high disconnected line.
Can you please help me? Thank you!