ncd-io/ncd-red-ads1115 #7: Having troubles managing internal MUX

Barabba11, 2019-01-17:
Hi there! I need again your help!
So as soon I connected a trimmer to every channel to test it I noticed that the module output channel 1 is connected to physical ADC3 input, and it displays about the same result (some time equal, some dime it differs from some units) to module output 4. Channels 2 and 3 output the same values, of channel ADC1.
This may be generated from wrong settings of MUX inside ADS1115, in your module I can configure address of mux and its port, I don't know what these datas are useful to, I've read in ADS1115 datasheet, the mux can be configured from config register as well as other functions existing in your module such comparator and so on, ADS1115 has not other I2C address. IN MUX is possible to choose I2C address as 112 to 119, and port, is is an eventually external MUX too? In this case I've to ignore it, but the settings "single ended" where I suppose to configure the internal mux I have random effects, for example:
I enable all inputs "single end", I have low voltage on ADC3, and I see the output ar ch4. If I disable the channel 2 I don't see more low voltage pin, ch1,3,4 have high disconnected line.
Can you please help me? Thank you!

In normal situation (all lines "single end") I have ADC1 who affects ch2 and ch3 with same value, ADC2 ignored, ADC3 who affects ch1 and ch4 with slight differences, ADC4 ignored

Seeing datasheet is not clear if you can set these register only first time and forget about them.. it looks before acquire a value they need to be set according the pin to read:
000 : AINP= AIN0 and AINN= AIN1(default)
001 : AINP= AIN0and AINN= AIN3
010 : AINP= AIN1and AINN= AIN3
011 : AINP= AIN2and AINN= AIN3
100 : AINP= AIN0and AINN= GND
101 : AINP= AIN1and AINN= GND
110 : AINP= AIN2and AINN= GND
111 : AINP= AIN3 and AINN= GND

for read pure ADC0 (that I named before ADC1) we need to set mux before to bit 100, for ADC1 101, etc.

Barabba11, 2019-02-01:
Hi there, any news? Thank you!

zaphod-42, 2019-02-01:
The configuration settings are sent to the chip every time a read is performed based on on your inputs, I've validated that everything is working on my end, I'll see if an engineer can take a look at this issue since I'm primarily a programmer and don't have the depth of understanding to help troubleshoot your setup any further.

Barabba11, 2019-02-03:
Thank you very much for your answer! As a programmer I would ask you, can the change I made affest this somehow? I edited just the html file and where you put addresses choice, I force to chose the address to 74.. I think this should not have impact with the script, but I ask eventullay.. 
It's very strange because as I wrote, using the library of Arduino and compiling it with RaspberryArduno to .bin file I had the correct 4 channells output. There should be something wrong in the script in my opinion..

zaphod-42, 2019-02-04:
Can you please post an export of your flow so I can check it out?

Barabba11, 2019-04-15:
Hi, sorry for delay and thanks for help. I've noticed that changing the rate/SPS gives also different results in outputs. For example now I have the channel 3 that outputs at ch1 and ch 4 (same results), and the ch1 which outputs on ch2 and ch 3. If I put the maximum sps 860 I have ch1 and ch2 with same results and ch3 and ch4 with same results. Amazing. Maybe this code was made for somthing else and importad in Raspberry? 

[{"id":"fa8a38aa.82a28","type":"ncd-ads1115","z":"fef64de3.66c588","name":"ADC 4 canali","connection":"2ab606d1.4ab30a","addr":"74","interval":"300","outputs":5,"output_all":true,"output_mult":"1","channel_1":"4","channel_2":"5","channel_3":"6","channel_4":"7","gain":"0","mode":"1","rate":"0","compMode":"1","compPol":"0","compLat":"0","compQueue":"3","delay":"100","x":410,"y":240,"wires":[["143ee412.ce898c"],["76299a7.5420ce4"],["17d15c27.14f654"],["d1bb1224.c0a29"],[]]},{"id":"826c9f43.97fdc","type":"LCD20x4-I2C","z":"fef64de3.66c588","name":"","speed":"2","size":"20x4","x":710,"y":220,"wires":[]},{"id":"143ee412.ce898c","type":"function","z":"fef64de3.66c588","name":"ADC1","func":"ADC1 = msg.payload.toString();\nglobal.set('ADC1',ADC1);\n","outputs":1,"noerr":0,"x":640,"y":360,"wires":[[]]},{"id":"69555800.b36ea8","type":"function","z":"fef64de3.66c588","name":"Gestisci LCD","func":"msga = {};\nvar valoreADC1 = global.get(\"ADC1\");\nvar valoreADC2 = global.get(\"ADC2\");\nvar valoreADC3 = global.get(\"ADC3\");\nvar valoreADC4 = global.get(\"ADC4\");\nmsg = {payload: [valoreADC1,valoreADC2,valoreADC3,valoreADC4]};\n\nvaloreADC1234 = valoreADC1+\" \"+valoreADC2+\" \"+valoreADC3;\nvar Tenda = global.get(\"TendaCamera\");\n\nvar IO0_AStr = global.get(\"IO0_AStr\");\nvar IO0_BStr = global.get(\"IO0_BStr\");\nvar IO0Str= \"A:\"+IO0_AStr+\"B:\"+IO0_BStr;\n\nvar IO1_AStr = global.get(\"IO1_AStr\");\nvar IO1_BStr = global.get(\"IO1_BStr\");\nvar IO1Str= \"A:\"+IO1_AStr+\"B:\"+IO1_BStr;\n\nmsga.payload = {msgs: [{msg: valoreADC1234 , pos:1,},{msg: valoreADC4,pos: 1,},{msg:  IO0Str,pos: 1,},{msg: IO1Str ,pos: 1,}]};\n\nreturn msga;\n","outputs":1,"noerr":0,"x":830,"y":540,"wires":[["826c9f43.97fdc"]]},{"id":"76299a7.5420ce4","type":"function","z":"fef64de3.66c588","name":"ADC2","func":"ADC2 = msg.payload.toString();\nglobal.set('ADC2',ADC2);\n","outputs":1,"noerr":0,"x":640,"y":420,"wires":[[]]},{"id":"17d15c27.14f654","type":"function","z":"fef64de3.66c588","name":"ADC3","func":"ADC3 = msg.payload.toString();\nglobal.set('ADC3',ADC3);\n","outputs":1,"noerr":0,"x":660,"y":480,"wires":[[]]},{"id":"d1bb1224.c0a29","type":"function","z":"fef64de3.66c588","name":"ADC4","func":"ADC4 = msg.payload.toString();\nglobal.set('ADC4',ADC4);\n\nreturn msg;","outputs":1,"noerr":0,"x":660,"y":540,"wires":[["69555800.b36ea8"]]},{"id":"2ab606d1.4ab30a","type":"ncd-comm","z":"","name":"We","bus":"i2c-1","commType":"standard","addr":"0","useMux":false,"muxAddr":"112","muxPort":"0"}]

Barabba11, 2019-04-16:
With NR 0,20 it started working only sometimes.. no output, please check it, thaks

Brianckramer, 2019-04-18:
I am seeing the same behavior, many times channels will show the exact same value, when I know they are not.  Re-Deploying sometimes helps, but not always.

zaphod-42, 2019-04-18:
@Barabba11 The library was written in NodeJS to minimize concerns about platform differences, that shouldn't be the issue.

 I did however find a problem in the code that could be causing this issue. I'll get a new test setup together as soon as I can to try it out. If you have time and would like to test out the change, it's pretty minor... in the ncd-ads1115.js file, change line 53 from:

`node.sensor.getSingleShot(channels[i]).then().catch();`

to:

`node.sensor.getSingleShot(channels[i], (i - 3) == 3).then().catch();`

and then line 121 from:

`node.sensor.getSingleShot(channels[chnl]).then((res) => {`

to:

`node.sensor.getSingleShot(channels[chnl], (chnl - 3) == 3).then((res) => {`

It looks like without this change it may be taking the readings in differential mode, which could account for the weirdness you are both experiencing. I'm currently in the middle of another project so like I said, I'll get a setup for test together as soon as I can, but I can't make the update to the package without verifying it actually works, so if either (or both) of you have time and wouldn't mind, we might be able to get this sorted much quicker with your help.

Thanks!

Barabba11, 2019-04-18:
thank you for support, I would underline that here we have 2 problems, one is reading the same values on other channels which I reported in past, the second one appeared after update to 0,20 and is that the module doesn't produce output, only sometimes after deploy.. it works tille the next deploy.. maybe, maybe not. I changed singleshot with auto, doesn't change. I tried to inject a get_status as you suggested but no answer too.

I would like to make your changes, just let me ask you:
Can I edit this JS without experience problems after in NR? there isn't any control of .js size and/or date of file?
Better to stop NR before edit and save them? 
To load the changes it needs to stop and start NR? or need to recompile the module somehow?
Kindly explain how to do the changes, thank you

Barabba11, 2019-04-22:
Hi, did you tried your module with the latest NR? it doesn't output unfortunately, please check, thanks

zaphod-42, 2019-04-22:
Running node version 11.4.0 and Node-RED version 0.20.5 I have no issues, output seems steady even between deploys. I have an engineer who is going to get me what I need to test the other issue today.

Yes, you can edit the package directly, it should be in ~/.node-red/node_modules/ncd-red-ads1115

Once the package has been changed you'll need to stop and start node-red, it won't affect any future upgrades to the package.

Barabba11, 2019-04-23:
Well, probably my mistake, sorry, miscobination of settings inside module, and changeing wiring of flow I created a wrong one.
I suppose today you ay release a new update with the fix, I'll wait, if you don't I'll try. Can you please include address 74 in your module too? I set it hardware and I can't change it, thank you!

zaphod-42, 2019-04-23:
Ok, I got the new version pushed (1.2.6), I did add address 74 to the dropdown and confirmed that the change I made fixed the error for single ended redings of multiple channels, please let me know if either of you have any more problems!

Barabba11, 2019-04-23:
Thank you a lot! tomorrow I'll check it

Barabba11, 2019-04-25:
Hi! There is no update 1.2.6 on palette so I forced overwrite 3 files under the module folder. Now I have this on startup, and NR hangs there:
25 Apr 10:38:51 - [info] Waiting for missing types to be registered:
25 Apr 10:38:51 - [info]  - ncd-ads1115
so I tried to recompile all:
cd ~/.node-red
npm rebuild

Still stuck there. Now my node red doesn't work more ((

Barabba11, 2019-04-25:
I delete it from flow, try to reinstall the module, but I can't uninstall it! It says me "failed to remove", the log is:

-----------------------------------------------------------
2019-04-25T07:45:00.435Z Remove : ncd-red-ads1115

-----------------------------------------------------------
2019-04-25T07:45:31.782Z Remove : ncd-red-ads1115

-----------------------------------------------------------
2019-04-25T07:46:19.831Z Remove : ncd-red-ads1115

Barabba11, 2019-04-25:
I understtod what was wrong, I downloaded only the 3 files you updated from github and the file .js for some reasons Firefox saved it on html and foratted it as html. So I renamed it and uploaded, but was a real html after that. So I downloaded zip, overwrite, uninstalled, reinstalled, overwrite again. Now 2 problems:
1) on Node red palette still 1.2.5 version, you need to ovewwtrite manually
2) Yes now it appears 74 (thanks for that) but still use 73, if it set 74 and I deploy, I open againa nd see 73, and ADS is disconnected of course

Barabba11, 2019-04-25:
I also have this on log now:
(node:23635) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
(node:23635) UnhandledPromiseRejectionWarning: Error: EREMOTEIO: remote I/O error, write
I didn't touch anythig esle in NR, so it should be that

Barabba11, 2019-04-25:
I made some experiments, lost some time, then I got that you made a mistake here:
	        <option value="72">74</option>
		<option value="73">74</option>
                <option value="73">74</option>

Now it works but no differences at outputs, I still have at output 1 and 4 the channel 3, and 2 and 3 the ch 1. As you said I stop node red, delete files, uploade them again, restart node red

Barabba11, 2019-04-25:
also a reboot didn't help, maybe I've to rebuild nodes? (npm rebuild)
I' sure I'm running the last .js because I opened it from RB and  for example
node.sensor.getSingleShot(channels[chnl], (chnl - 3) == 3).then((res) => {

daxkamala, 2020-03-29:
Hi. I've using this for a long time without any issues, very impressed. Sadly, I seem to have run into the problems of the node not starting until a number of re-deploys after which it runs fine until the next deploy. It started when I upgraded to Node-Red v1.0.4. ncd-ads115 is 1.2.7. Reading back it seems the problem was fixed. Any hints on why I'm now getting this? Thx