ncd-io/ncd-red-ads1115 #8: Node isn't closing correctly.

Proamp, 2019-07-01:
If I add a subflow with ADC node in flow, the first time it works fine, but after I remove the subflow and deploy, ADC still works.

Further - worse. If I add and remove subflow ADC second time, I see 2 ADC works simultaneously without any ADC in flow.

I made a small video with timeline comments to show the problem:
https://www.youtube.com/watch?v=nRQSKSKsp5Q

Proamp, 2019-07-01:
Also a little typo with i2c address 74
[image]

Barabba11, 2019-07-01:
Yes there is a small issue there, but don't have a clue with the problem you are complaining.
On adc is possible to set an auto update every x milliseconds or set it to work on a trigger, which is a particular msg.payload, I don't remember now. If you don't select that it's normal that art keeps converting automatically. The only way to stop flow is by the prompt command node-red-stop. When you deploying you are not stopping it, just a fraction of second.
I have a question, what adc results you have? I'm suffering from a mismatch of channels, k see for example at output 1 and 4 the channel 3 and at output 2 and 3 the channel 1, or something like that. Do you have everything in order? You tried to connect a trimmer to every channel? Thanks