ncd-io/ncd-red-ads1115 #9: i2c connection field is empty and doesn't show dropdown menu

jarcher12, 2019-12-28:
i have uninstalled and reinstalled the node, i have a brand new installation of raspian and node red. also installed the ncd-red-comm nodes seperately . Do you havge any ideas ?
[image]

jarcher12, 2019-12-28:
this throws the error 
12/28/2019, 7:14:59 PMnode: e8651b4d.ab7e68
msg : error
"TypeError: Cannot read property 'i2c' of null"

[image]

jarcher12, 2019-12-28:
got one step further to the edit portion of the ncd-red-comm node but the i2c bus still wont populate
[image]

jarcher12, 2019-12-28:
Looks to be an i2c issue unrelated to this particular node

jarcher12, 2019-12-29:
I reinstalled buster but to no avail. this is the install log for the node. it seems a lot of stuff is broken.

-----------------------------------------------------------
2019-12-29T09:21:05.885Z Install : ncd-red-ads1115 1.2.7

2019-12-29T09:21:05.858Z npm install --no-audit --no-update-notifier --save --save-prefix="~" --production ncd-red-ads1115@1.2.7
2019-12-29T09:21:21.801Z [out] 
2019-12-29T09:21:21.801Z [out] > i2c-bus@3.2.0 install /home/pi/.node-red/node_modules/i2c-bus
2019-12-29T09:21:21.801Z [out] > node-gyp rebuild
2019-12-29T09:21:21.801Z [out] 
2019-12-29T09:21:24.456Z [out] make: Entering directory '/home/pi/.node-red/node_modules/i2c-bus/build'
2019-12-29T09:21:24.462Z [out]   CXX(target) Release/obj.target/i2c/src/i2c.o
2019-12-29T09:21:28.202Z [err] In file included from ../src/i2c.cc:2:
2019-12-29T09:21:28.202Z [err] ../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
2019-12-29T09:21:28.202Z [err] ../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
2019-12-29T09:21:28.202Z [err]      , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
2019-12-29T09:21:28.202Z [err]                                                               ^
2019-12-29T09:21:28.315Z [err] In file included from ../src/i2c.cc:1:
2019-12-29T09:21:28.315Z [err] ../src/i2c.cc: At global scope:
2019-12-29T09:21:28.315Z [err] /home/pi/.cache/node-gyp/12.14.0/include/node/node.h:566:43: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
2019-12-29T09:21:28.315Z [err]        (node::addon_register_func) (regfunc),                          \
2019-12-29T09:21:28.315Z [err]                                            ^
2019-12-29T09:21:28.315Z [err] /home/pi/.cache/node-gyp/12.14.0/include/node/node.h:600:3: note: in expansion of macro ‘NODE_MODULE_X’
2019-12-29T09:21:28.315Z [err]    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
2019-12-29T09:21:28.315Z [err]    ^~~~~~~~~~~~~
2019-12-29T09:21:28.315Z [err] ../src/i2c.cc:92:1: note: in expansion of macro ‘NODE_MODULE’
2019-12-29T09:21:28.315Z [err]  NODE_MODULE(i2c, InitAll)
2019-12-29T09:21:28.315Z [err]  ^~~~~~~~~~~
2019-12-29T09:21:28.330Z [err] In file included from ../src/i2c.cc:99:
2019-12-29T09:21:28.330Z [err] ../src/./i2cfuncs.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE I2cFuncsAsync(Nan::NAN_METHOD_ARGS_TYPE)’:
2019-12-29T09:21:28.330Z [err] ../src/./i2cfuncs.cc:48:32: error: no matching function for call to ‘v8::Value::Int32Value()’
2019-12-29T09:21:28.330Z [err]    int fd = info[0]->Int32Value();
2019-12-29T09:21:28.330Z [err]                                 ^
2019-12-29T09:21:28.330Z [err] In file included from /home/pi/.cache/node-gyp/12.14.0/include/node/node.h:63,
2019-12-29T09:21:28.330Z [err]                  from ../src/i2c.cc:1:
2019-12-29T09:21:28.330Z [err] /home/pi/.cache/node-gyp/12.14.0/include/node/v8.h:2613:40: note: candidate: ‘v8::Maybe<int> v8::Value::Int32Value(v8::Local<v8::Context>) const’
2019-12-29T09:21:28.330Z [err]    V8_WARN_UNUSED_RESULT Maybe<int32_t> Int32Value(Local<Context> context) const;
2019-12-29T09:21:28.330Z [err]                                         ^~~~~~~~~~
2019-12-29T09:21:28.330Z [err] /home/pi/.cache/node-gyp/12.14.0/include/node/v8.h:2613:40: note:   candidate expects 1 argument, 0 provided
2019-12-29T09:21:28.332Z [err] In file included from ../src/i2c.cc:99:
2019-12-29T09:21:28.332Z [err] ../src/./i2cfuncs.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE I2cFuncsSync(Nan::NAN_METHOD_ARGS_TYPE)’:
2019-12-29T09:21:28.332Z [err] ../src/./i2cfuncs.cc:60:32: error: no matching function for call to ‘v8::Value::Int32Value()’
2019-12-29T09:21:28.332Z [err]    int fd = info[0]->Int32Value();
2019-12-29T09:21:28.332Z [err]                                 ^
2019-12-29T09:21:28.333Z [err] In file included from /home/pi/.cache/node-gyp/12.14.0/include/node/node.h:63,
2019-12-29T09:21:28.333Z [err]                  from ../src/i2c.cc:1:
2019-12-29T09:21:28.333Z [err] /home/pi/.cache/node-gyp/12.14.0/include/node/v8.h:2613:40: note: candidate: ‘v8::Maybe<int> v8::Value::Int32Value(v8::Local<v8::Context>) const’
2019-12-29T09:21:28.333Z [err]    V8_WARN_UNUSED_RESULT Maybe<int32_t> Int32Value(Local<Context> context) const;
2019-12-29T09:21:28.333Z [err]                                         ^~~~~~~~~~
2019-12-29T09:21:28.333Z [err] /home/pi/.cache/node-gyp/12.14.0/include/node/v8.h:2613:40: note:   candidate expects 1 argument, 0 provided
2019-12-29T09:21:28.342Z [err] In file included from ../src/i2c.cc:100:
2019-12-29T09:21:28.342Z [err] ../src/./deviceid.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE DeviceIdAsync(Nan::NAN_METHOD_ARGS_TYPE)’:
2019-12-29T09:21:28.342Z [err] ../src/./deviceid.cc:49:32: error: no matching function for call to ‘v8::Value::Int32Value()’
2019-12-29T09:21:28.342Z [err]    int fd = info[0]->Int32Value();
2019-12-29T09:21:28.342Z [err]                                 ^
2019-12-29T09:21:28.344Z [err] In file included from /home/pi/.cache/node-gyp/12.14.0/include/node/node.h:63,
2019-12-29T09:21:28.344Z [err]                  from ../src/i2c.cc:1:
[... cut ...]
2019-12-29T09:21:52.703Z [err] npm WARN 
2019-12-29T09:21:52.703Z [err] optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
2019-12-29T09:21:52.704Z [err] 
2019-12-29T09:21:52.708Z [out] + ncd-red-ads1115@1.2.7
2019-12-29T09:21:52.708Z [out] added 14 packages from 2 contributors in 43.933s
2019-12-29T09:21:53.157Z [out] 
2019-12-29T09:21:53.157Z [out] 2 packages are looking for funding
2019-12-29T09:21:53.157Z [out]   run `npm fund` for details
2019-12-29T09:21:53.157Z [out] 
2019-12-29T09:21:53.196Z rc=0

nate8199, 2020-02-02:
I had this issue also, had to downgrade node to v8 and reinstall the the i2c-bus and ncd nodes to get it to show up.

m4c1n0, 2020-02-12:
Hi I had the same issue that I wasn't able to select anything on the I2C Bus drop-down.

Following [this guide](https://ozzmaker.com/i2c/) I installed **i2c-tools**, **libi2c-dev** and **python-smbus** as an attempt to troubleshoot if the RPi itself sees the ADS-1115 sensor.

After installing and rebooting I issued the command:

> sudo i2cdetect -y 1

to see if the sensor is present and visible. It was visible, so I went back to node-red to see if anything changed and suddenly I was able to select i2c-1 as the "I2C Bus".

Hope this helps.

RogierQ, 2020-02-13:
Same here. but not able to solve it with the solution above

zaphod-42, 2020-03-13:
This issue should be solved with a reinstall, the i2c-bus library was updated to support newer node versions, the ncd-red-comm dependency of this node has been updated accordingly.

closed 2020-03-13 by zaphod-42

justinSCS, 2020-09-11:
@zaphod-42 I am running node red in a container and running into the same issue, dropdown is empty and no ability to add a bus.  I can run i2cdetect -y 1 and see the NCD module on my system.  I have tried:

-installing the ncd-red-comm library as a separate line from the ncd-red-ads1115 in my dockerfile, 
-manually installing the ncd-red-comm through npm after the container is up and running
-manually installing another ncd node from the pallete manager in node red after the container starts (in hope of inducing any corrections on the docker install if there were any)
-grabbing a node flow from someone elses post that included the ncd-red-comm node as well as the ads1115 node in hopes of forcing the comm node to be made (I got an error for unrecognized node ncd-comm on node-red interface(sidenote "type":"ncd-comm", from actual json)
-installing i2c-bus in docker as well as updating manually
-deleteing ads1115 node from pallete and reinstalling from palette manager manually

Is there something I might be missing or be doing wrong?

justinSCS, 2020-09-11:
Update:
I cleared my dockerfile of NCD nodes as well as cleared my flow of them, pushed an updated container and manually installed ncd-erd-ads1115 manually again and it worked fine.

I reinserted the ads1115 node back into my dockerfile and added the ads1115 node (with the comm node created) in my flow and pushed the updated container. Everything downloaded and seems to be working fine.  I have no explanation